HKUST-Aerial-Robotics/open_quadtree_mapping #3: Can't read in

hitzhang, 2018-06-16:
use VINS publish the two topics
but it is not starting.
 i have modified the parameters

WANG-KX, 2018-06-17:
The original VINS－MONO publishes camera poses in nav_msgs/Odometry. Here we subscribe the message in geometry_msgs/PoseStamped.
Please modify one of the code so that you can subscribe.
PS, If you are modifying open_quadtree_mapping, please see src/main_ros.cpp, src/depthmap_node.cpp, and include/quadmap/depthmap_node.h.

closed 2018-06-17 by WANG-KX

hitzhang, 2018-06-17:
should i modify const eometry_msgs::/PoseStampedConstPtr &pose_input) to const nav_msgs::OdometryConstPtr &pose_input) in src/depthmap_node.cpp?

there is debug error.
/home/zhang/azhang/mapmap/src/open_quadtree_mapping/src/depthmap_node.cpp: In member function ‘void quadmap::DepthmapNode::Msg_Callback(const ImageConstPtr&, const OdometryConstPtr&)’:
/home/zhang/azhang/mapmap/src/open_quadtree_mapping/src/depthmap_node.cpp:131:26: error: ‘const _pose_type {aka const struct geometry_msgs::PoseWithCovariance_<std::allocator<void> >}’ has no member named ‘orientation’
         pose_input->pose.orientation.w,
                          ^
/home/zhang/azhang/mapmap/src/open_quadtree_mapping/src/depthmap_node.cpp:132:26: error: ‘const _pose_type {aka const struct geometry_msgs::PoseWithCovariance_<std::allocator<void> >}’ has no member named ‘orientation’
         pose_input->pose.orientation.x,
                          ^
/home/zhang/azhang/mapmap/src/open_quadtree_mapping/src/depthmap_node.cpp:133:26: error: ‘const _pose_type {aka const struct geometry_msgs::PoseWithCovariance_<std::allocator<void> >}’ has no member named ‘orientation’
         pose_input->pose.orientation.y,
                          ^
/home/zhang/azhang/mapmap/src/open_quadtree_mapping/src/depthmap_node.cpp:134:26: error: ‘const _pose_type {aka const struct geometry_msgs::PoseWithCovariance_<std::allocator<void> >}’ has no member named ‘orientation’
         pose_input->pose.orientation.z,
                          ^
/home/zhang/azhang/mapmap/src/open_quadtree_mapping/src/depthmap_node.cpp:135:26: error: ‘const _pose_type {aka const struct geometry_msgs::PoseWithCovariance_<std::allocator<void> >}’ has no member named ‘position’
         pose_input->pose.position.x,
                          ^
/home/zhang/azhang/mapmap/src/open_quadtree_mapping/src/depthmap_node.cpp:136:26: error: ‘const _pose_type {aka const struct geometry_msgs::PoseWithCovariance_<std::allocator<void> >}’ has no member named ‘position’
         pose_input->pose.position.y,
                          ^
/home/zhang/azhang/mapmap/src/open_quadtree_mapping/src/depthmap_node.cpp:137:26: error: ‘const _pose_type {aka const struct geometry_msgs::PoseWithCovariance_<std::allocator<void> >}’ has no member named ‘position’
         pose_input->pose.position.z);
                          ^
make[2]: *** [open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/src/depthmap_node.cpp.o] 错误 1
make[1]: *** [open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/all] 错误 2
make: *** [all] 错误 2
Invoking "make -j8 -l8" failed

WANG-KX, 2018-06-17:
It is not that simple.
Additionally, you should include`#include <sensor_msgs/PointCloud.h>` and the way to access pose data is different.
Please refer to [Odometry](http://docs.ros.org/kinetic/api/nav_msgs/html/msg/Odometry.html).
for example:
`  quadmap::SE3<float> T_world_curr(
        pose_input->pose.pose.orientation.w,
        pose_input->pose.pose.orientation.x,
        pose_input->pose.pose.orientation.y,
        pose_input->pose.pose.orientation.z,
        pose_input->pose.pose.position.x,
        pose_input->pose.pose.position.y,
        pose_input->pose.pose.position.z);`

hitzhang, 2018-06-17:
Worked. Thanks a lot.

slamlixiaodong, 2019-03-07:
> Worked. Thanks a lot.

How did you modify the code to make it worked? I encountered the same trouble. I hope you can give me some advice.

WANG-KX, 2019-03-07:
Here, I provide an example to convert.
[odom2pose.tar.gz](https://github.com/HKUST-Aerial-Robotics/open_quadtree_mapping/files/2939602/odom2pose.tar.gz)

slamlixiaodong, 2019-03-07:
> Here, I provide an example to convert.
> [odom2pose.tar.gz](https://github.com/HKUST-Aerial-Robotics/open_quadtree_mapping/files/2939602/odom2pose.tar.gz)

Worked. Thank you very much.