HKUST-Aerial-Robotics/open_quadtree_mapping #4: No rule to make target '/usr/lib/x86_64-linux-gnu/libvtkproj4-6.2.so.6.2.0'

TianQi-777, 2018-07-31:
[ 85%] Building CXX object open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/src/main_ros.cpp.o

make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libvtkproj4-6.2.so.6.2.0', needed by '/home/tq/a2_demo/dense_vins/devel/lib/open_quadtree_mapping/open_quadtree_mapping_node'。 停止。
make[2]: *** 正在等待未完成的任务....
/home/tq/a2_demo/dense_vins/src/open_quadtree_mapping/src/depthmap.cpp: In constructor ‘quadmap::Depthmap::Depthmap(std::size_t, std::size_t, float, float, float, float, cv::Mat, cv::Mat, int)’:
/home/tq/a2_demo/dense_vins/src/open_quadtree_mapping/src/depthmap.cpp:39:110: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘std::size_t {aka long unsigned int}’ [-Wformat=]
 d (%d x %d) fx: %f, fy: %f, cx: %f, cy: %f.\n", width, height, fx, fy, cx, cy);
                                                                              ^
/home/tq/a2_demo/dense_vins/src/open_quadtree_mapping/src/depthmap.cpp:39:110: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘std::size_t {aka long unsigned int}’ [-Wformat=]
CMakeFiles/Makefile2:938: recipe for target 'open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/all' failed
make[1]: *** [open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Does this demo have a special requirement for openCV? I use openCV3.4. How should I solve this problem?

WANG-KX, 2018-07-31:
If you are not familiar with OpenCV compiled with cuda, I suggest you use no_cudaopencv branch.
It seems to be the libvtk problem. In my system, I used PCL 1.8 to fuse the depth maps. You can install the PCL 1.8 following [OpenChisel](https://github.com/personalrobotics/OpenChisel) under "A note on PCL".
Hope this can solve the problem.

TianQi-777, 2018-07-31:
Thank you for your prompt reply. I also feel PCL has some problems. My current PCL version is 1.7. I'll try it again with 1.8.

TianQi-777, 2018-07-31:
“Quadtree-accelerated Real-time Monocular Dense Mapping” Is this paper open now? I can't find the link.

WANG-KX, 2018-07-31:
The paper is accepted at IROS 2018. It will be released after the conference.

WANG-KX, 2018-08-01:
I think you forgot to source the ROS environment, e.g. source ~/catkin_ws/devel/setup.bash.
You can find the correct way to run ROS at http://wiki.ros.org/ROS/Tutorials.
run"./devel/lib/open_quadtree_mapping/open_quadtree_maing_node" is not correct because you did not pass any parameters into the node.

WANG-KX, 2018-08-01:
Or try

> rospack profile

TianQi-777, 2018-08-01:
path:        dense_output.bag
version:     2.0
duration:    3:48s (228s)
start:       Aug 01 2018 13:40:37.94 (1533102037.94)
end:         Aug 01 2018 13:44:26.03 (1533102266.03)
size:        4.1 GB
messages:    2963
compression: none [1519/1519 chunks]
types:       geometry_msgs/PoseStamped [d3812c3cbc69362b77dc0b19b345f8f5]
             sensor_msgs/Image         [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/PointCloud2   [1158d486dd51d683ce2f1be655c3c181]
topics:      /mv_25001498/image_raw               685 msgs    : sensor_msgs/Image        
             /open_quadtree_mapping/color_depth   376 msgs    : sensor_msgs/Image        
             /open_quadtree_mapping/debug         376 msgs    : sensor_msgs/Image        
             /open_quadtree_mapping/depth         376 msgs    : sensor_msgs/Image        
             /open_quadtree_mapping/pointcloud    360 msgs    : sensor_msgs/PointCloud2  
             /open_quadtree_mapping/reference     376 msgs    : sensor_msgs/Image        
             /vins_estimator/camera_pose          414 msgs    : geometry_msgs/PoseStamped

This is my final output, Why is pointcloud not compatible with other output?
pointcloud(360 msgs)
color_depth(376 msgs) 
debug(376 msgs)
reference(376 msgs)

TianQi-777, 2018-08-01:
I want to use PCL to display the output point cloud and see the effect. I need to know if the coordinates of the point cloud are world coordinates or camera coordinates?

WANG-KX, 2018-08-01:
First question: the difference between master and no_cudaopencv is that the former one undistort images using opencv cuda functions. the OpenCV must be compiled with CUDA.

Second question: the system needs several frams to initialize (to get enough measurement frames). After the initialization, depth maps are generated for every input image.

WANG-KX, 2018-08-01:
Point clouds are in the World frame.

TianQi-777, 2018-08-01:
Okay, this is a very good project, All my problems are solved, I will close this issue, thank you for your prompt reply :)

closed 2018-08-01 by TianQi-777