HKUST-Aerial-Robotics/open_quadtree_mapping #6: example.bag download

niannianmeng, 2018-08-03:
is the mainland China can access to the link for the example.bag?I tried, but failed

WANG-KX, 2018-08-03:
Hello,
The .bag file is hosted at Dropbox and may be difficult for mainland China to access. I would like to host the file under the cloud you can access. Please tell me which cloud works best in mainland China? As far as I know, Baidu Yun may be too slow.

WANG-KX, 2018-08-03:
Hello,
I have uploaded the file to OneDrive, please try this [link](https://1drv.ms/u/s!Av7Og_26nifdpwTQKNt6OhX5X2ed). Does it work well?

niannianmeng, 2018-08-03:
Hi, KX,
thanks a lot for you reply, and try to upload another cloud OneDrive for me. but i still can not access to OneDrive link. as far as I know, Baidu yun is ok for me, is it too slow for you uplode file?

WANG-KX, 2018-08-03:
Please try this [link](https://pan.baidu.com/s/17LcpYUflgBtoSaMskmWWDw)

niannianmeng, 2018-08-03:
yes, it is ok, thank you very much

closed 2018-08-03 by WANG-KX