HKUST-Aerial-Robotics/open_quadtree_mapping #8: Parameter tuning towards better quality

Changliu52, 2018-09-13:
Hi,
Thank you so much for sharing this awesome code.
I wonder if you have any suggestions on how to tune the parameters (both in launch file and stereo_parameter.cuh) in order to obtain higher quality pointclouds, when running in a powerful machine.

currently it is kind of wave towards, and very dependent on the motion. 

Thank you very much in advance.
Chang

WANG-KX, 2018-09-13:
Dear,

Thanks for your interest in our work!

The quality of monocular dense depth estimation depends on the movement to some extent.
To get better results, my experience is that a good camera calibration and pose estimation are required.  Also, too much rotation is not good that each pixel may not have enough observations. We did not fine-tune the parameters in the code too much. Basically,  you can follow section 3.0 of the README.md. Most of the parameters in "stereo_parameter.cuh" are from LSD-SLAM to calculate the semi-dense maps on GPU and we did not tune them.

During the development, we evaluated the open_quadtree_mapping using lots of datasets. We find that as long as you have a good calibration, pose estimation and small rotations, the quality is acceptable.

If you want a better monocular dense mapping approach for your applications and do not care about the computation burden, I think you can try [MVDepthNet](https://github.com/HKUST-Aerial-Robotics/MVDepthNet). Recently, we find that the depth from MVDepthNet is denser than that from open_quadtree_mapping, and is robust to camera calibrations, reflective surfaces etc. However, MVDepthNet requires a much more powerful GPU to run.

I hope this helps you.

Changliu52, 2018-09-13:
Hi Kaixuan,

Thank you so much for the interesting reply.
How easy to set a threshold on how much the pose translation is required for each frame of point cloud? currently it outputs noisy point cloud when the camera is stationary.

Besides, maybe there is a parameter to set the threshold for the covariance for the points to be added to the point cloud, so that points that didn’t get enough observation won’t be added to the point cloud. The parameter VARIANCE_MAX in stereo_parameter.cuh seems to do this but doesn’t affect much when I decrease it to 0.001. Maybe I was doing something completely nonsense?

Best Regards,
Chang

WANG-KX, 2018-09-15:
Dear,

We may go deep in the code.

We use a simple strategy to select the keyframes for depth estimation. Basically, a rotation and a translation threshold are used to filter out nearby poses. Ideally, stationary poses can be handled if the odometry system does not drift in time. For the keyframe selection, you can check [this function](https://github.com/HKUST-Aerial-Robotics/open_quadtree_mapping/blob/eac9f2f0c8a7bf8b368bd60d97a2256182bf4c7a/src/seed_matrix.cu#L200).

The published point cloud is only the points projected from **this frame**, no history point clouds are stored. The reason that the point cloud maybe not so much is that we use a frame to frame filter to check inlier points. The filter is sometime similar to the depth filter of SVO. Comment [this line](https://github.com/HKUST-Aerial-Robotics/open_quadtree_mapping/blob/eac9f2f0c8a7bf8b368bd60d97a2256182bf4c7a/src/seed_matrix.cu#L165) to disable the frame to frame filter. You can get 100% dense point clouds but they are really noisy if the input images are not good. **Personly I do not recommend to do this.** Or you can check the code of the filer by following the code of `fuse_output_depth()`. In shorts, [parameter a](https://github.com/HKUST-Aerial-Robotics/open_quadtree_mapping/blob/eac9f2f0c8a7bf8b368bd60d97a2256182bf4c7a/src/depth_fusion.cu#L13), [parameter b](https://github.com/HKUST-Aerial-Robotics/open_quadtree_mapping/blob/eac9f2f0c8a7bf8b368bd60d97a2256182bf4c7a/src/depth_fusion.cu#L13), [this function](https://github.com/HKUST-Aerial-Robotics/open_quadtree_mapping/blob/eac9f2f0c8a7bf8b368bd60d97a2256182bf4c7a/src/depth_fusion.cu#L21), and [this function](https://github.com/HKUST-Aerial-Robotics/open_quadtree_mapping/blob/eac9f2f0c8a7bf8b368bd60d97a2256182bf4c7a/src/depth_fusion.cu#L25) control the output of the filter.

Hope this can help you!

Regards,
Kaixuan

closed 2018-10-19 by WANG-KX