HKUST-Aerial-Robotics/open_quadtree_mapping #13: Connection error

stevechan1993, 2018-12-25:
When I try to connect to VINS Mono,I got these error:
[ERROR] [1545746670.569675978]: Client [/open_quadtree_mapping] wants topic /vins_estimator/camera_pose to have datatype/md5sum [geometry_msgs/PoseStamped/d3812c3cbc69362b77dc0b19b345f8f5], but our version has [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7]. Dropping connection.
What's the problem is that?

WANG-KX, 2018-12-27:
You need to convert the message type. In open_quadtree_mapping we use [PoseStamped](http://docs.ros.org/lunar/api/geometry_msgs/html/msg/PoseStamped.html) but VINS uses [Odometry](http://docs.ros.org/melodic/api/nav_msgs/html/msg/Odometry.html). You can change the code in open_quadtree_mapping.

stevechan1993, 2018-12-28:
> You need to convert the message type. In open_quadtree_mapping we use [PoseStamped](http://docs.ros.org/lunar/api/geometry_msgs/html/msg/PoseStamped.html) but VINS uses [Odometry](http://docs.ros.org/melodic/api/nav_msgs/html/msg/Odom

closed 2018-12-28 by stevechan1993