HKUST-Aerial-Robotics/open_quadtree_mapping #17: md5sum error

stevechan1993, 2019-02-15:
Running open chisel on a machine which have ros indigo, and try to subscribe the topic: /Chisel/full_mesh from a machine which have ros kinetic, got this error mesage like:
wants topic /Chisel/full_mesh to have datatype/md5sum [visualization_msgs/Marker/4048c9de2a16f4ae8e0538085ebf1b97], but our version has [visualization_msgs/Marker/18326976df9d29249efc939e00342cde]. Dropping connection.
Is there any way to communicate between different distribution of ros? Or can I use any other map fusion on ubuntu 16.04 with ros kinetic?

WANG-KX, 2019-02-15:
Well ...
The messages are not the same on different ROS versions. I do not know how to solve it unless using the same ROS version.
Maybe you can try other ROS message types, such as std_msgs/Float32MultiArray, and convert into meshes yourself.

stevechan1993, 2019-02-16:
Thank you for answering.Do you have any idea of convert visualization_msgs/Marker to sensor_msgs/PointCloud2?

WANG-KX, 2019-02-17:
It depends on what do you care in your applications.
If you only need obstacle information to do navigation, I think you can safely publish a point cloud with each point corresponding to the mesh location.
If you want the mesh information for 3D reconstruction, I think you need more effort to write the code. It all depends on how you pack the information into sensor_msgs/PointCloud2/data and decode it on the receiver computer. 
Hope it helps you.

stevechan1993, 2019-02-18:
Thank you for your help. Actually I only need to do navigation for now.

closed 2019-02-23 by WANG-KX