HKUST-Aerial-Robotics/open_quadtree_mapping #20: a catkin_make error

hengwangzhao, 2019-05-01:
ROS kinetic
CUDA 8.0
I have this error:

/home/zhw/catkin_ws/src/open_quadtree_mapping/src/check_cuda_device.cu:21:41: fatal error: quadmap/check_cuda_device.cuh: No such file or directory
compilation terminated.
/home/zhw/catkin_ws/src/open_quadtree_mapping/src/seed_matrix.cu:1:35: fatal error: quadmap/seed_matrix.cuh: No such file or directory
compilation terminated.
CMake Error at CMakeFiles/open_quadtree_mapping_node_generated_seed_matrix.cu.o.cmake:215 (message):
  Error generating
  /home/zhw/catkin_ws/build/open_quadtree_mapping/./open_quadtree_mapping_node_generated_seed_matrix.cu.o

CMake Error at CMakeFiles/open_quadtree_mapping_node_generated_check_cuda_device.cu.o.cmake:215 (message):
  Error generating
  /home/zhw/catkin_ws/build/open_quadtree_mapping/./open_quadtree_mapping_node_generated_check_cuda_device.cu.o

open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/build.make:62: recipe for target 'open_quadtree_mapping/open_quadtree_mapping_node_generated_check_cuda_device.cu.o' failed
make[2]: *** [open_quadtree_mapping/open_quadtree_mapping_node_generated_check_cuda_device.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/build.make:68: recipe for target 'open_quadtree_mapping/open_quadtree_mapping_node_generated_seed_matrix.cu.o' failed
make[2]: *** [open_quadtree_mapping/open_quadtree_mapping_node_generated_seed_matrix.cu.o] Error 1
CMakeFiles/Makefile2:7540: recipe for target 'open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/all' failed
make[1]: *** [open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

How can I solve it? Thank you!

hengwangzhao, 2019-05-01:
I solved it  by chaning  it to the Absolute path, like:
#include <quadmap / seed_matrix.cuh> to
 #include </home/zhw/catkin_ws/src/open_quadtree_mapping/include/quadmap/matrix.cuh>

But i have got another problem:

####
#### Running command: "make cmake_check_build_system" in "/home/zhw/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/zhw/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/zhw/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/zhw/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/zhw/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- Using CATKIN_WHITELIST_PACKAGES: open_quadtree_mapping
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - open_quadtree_mapping
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'open_quadtree_mapping'
-- ==> add_subdirectory(open_quadtree_mapping)
-- Eigen found (include: /usr/include/eigen3)
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zhw/catkin_ws/build
####
#### Running command: "make -j12 -l12" in "/home/zhw/catkin_ws/build"
####
[ 14%] Building NVCC (Device) object open_quadtree_mapping/open_quadtree_mapping_node_generated_seed_matrix.cu.o
[ 28%] Building NVCC (Device) object open_quadtree_mapping/open_quadtree_mapping_node_generated_check_cuda_device.cu.o
/home/zhw/catkin_ws/src/open_quadtree_mapping/src/depth_upsample.cu(329): warning: variable "keypoint_step" was declared but never referenced

/home/zhw/catkin_ws/src/open_quadtree_mapping/src/depth_upsample.cu(329): warning: variable "keypoint_step" was declared but never referenced

/home/zhw/catkin_ws/src/open_quadtree_mapping/src/seed_matrix.cu: In member function ‘void quadmap::SeedMatrix::extract_depth()’:
/home/zhw/catkin_ws/src/open_quadtree_mapping/src/seed_matrix.cu:456:57: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘std::deque<quadmap::FrameElement>::size_type {aka long unsigned int}’ [-Wformat=]
   printf("  we have %d frames.\n", framelist_host.size());
                                                         ^
Scanning dependencies of target open_quadtree_mapping_node
[ 42%] Building CXX object open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/src/depthmap.cpp.o
[ 57%] Building CXX object open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/src/depthmap_node.cpp.o
[ 71%] Building CXX object open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/src/publisher.cpp.o
[ 85%] Building CXX object open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/src/main_ros.cpp.o
/home/zhw/catkin_ws/src/open_quadtree_mapping/src/depthmap.cpp: In constructor ‘quadmap::Depthmap::Depthmap(std::size_t, std::size_t, float, float, float, float, cv::Mat, cv::Mat, int)’:
/home/zhw/catkin_ws/src/open_quadtree_mapping/src/depthmap.cpp:39:110: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘std::size_t {aka long unsigned int}’ [-Wformat=]
 d (%d x %d) fx: %f, fy: %f, cx: %f, cy: %f.\n", width, height, fx, fy, cx, cy);
                                                                              ^
/home/zhw/catkin_ws/src/open_quadtree_mapping/src/depthmap.cpp:39:110: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘std::size_t {aka long unsigned int}’ [-Wformat=]
[100%] Linking CXX executable /home/zhw/catkin_ws/devel/lib/open_quadtree_mapping/open_quadtree_mapping_node
open_quadtree_mapping_node_generated_seed_matrix.cu.o: In function `quadmap::SeedMatrix::set_income_as_keyframe()':
tmpxft_00004770_00000000-4_seed_matrix.cudafe1.cpp:(.text+0x1905): undefined reference to `cv::_OutputArray::_OutputArray(cv::Mat&)'
open_quadtree_mapping_node_generated_seed_matrix.cu.o: In function `quadmap::SeedMatrix::get_result(cv::Mat&, cv::Mat&, cv::Mat&)':
tmpxft_00004770_00000000-4_seed_matrix.cudafe1.cpp:(.text+0x2ca7): undefined reference to `cv::_OutputArray::_OutputArray(cv::Mat&)'
tmpxft_00004770_00000000-4_seed_matrix.cudafe1.cpp:(.text+0x2e3b): undefined reference to `cv::_OutputArray::_OutputArray(cv::Mat&)'
tmpxft_00004770_00000000-4_seed_matrix.cudafe1.cpp:(.text+0x2fb8): undefined reference to `cv::_OutputArray::_OutputArray(cv::Mat&)'
open_quadtree_mapping_node_generated_seed_matrix.cu.o: In function `quadmap::SeedMatrix::input_raw(cv::Mat, quadmap::SE3<float>)':
tmpxft_00004770_00000000-4_seed_matrix.cudafe1.cpp:(.text+0xc128): undefined reference to `cv::_InputArray::_InputArray(cv::Mat const&)'
tmpxft_00004770_00000000-4_seed_matrix.cudafe1.cpp:(.text+0xc139): undefined reference to `cv::_InputArray::_InputArray(cv::Mat const&)'
tmpxft_00004770_00000000-4_seed_matrix.cudafe1.cpp:(.text+0xc146): undefined reference to `cv::_OutputArray::_OutputArray(cv::Mat&)'
[... cut ...]
open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/build.make:432: recipe for target '/home/zhw/catkin_ws/devel/lib/open_quadtree_mapping/open_quadtree_mapping_node' failed
make[2]: *** [/home/zhw/catkin_ws/devel/lib/open_quadtree_mapping/open_quadtree_mapping_node] Error 1
CMakeFiles/Makefile2:943: recipe for target 'open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/all' failed
make[1]: *** [open_quadtree_mapping/CMakeFiles/open_quadtree_mapping_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

How can I fix it?

closed 2019-05-02 by hengwangzhao

WANG-KX, 2019-05-02:
Actually, I don't know how to solve it. It seems that some environment setting is not consistent with my settings. If you solved the problem, please post here.

Thanks a lot!

zhijianglu, 2020-05-26:
> Actually, I don't know how to solve it. It seems that some environment setting is not consistent with my settings. If you solved the problem, please post here.
> 
> Thanks a lot!

This package needs to be compiled first by:
       catkin_make -DCATKIN_WHITELIST_PACKAGES="open_quadtree_mapping" 
then modify the package name to compile other packages in turn