HKUST-Aerial-Robotics/open_quadtree_mapping #21: How to implement real-time dense mapping with VINS-MONO and quadtree_mapping?

scottymax, 2019-05-29:
Hello, I have successfully run VINS-MONO and open_quadtree_mapping programs respectively. Now I want  to combine VINS-MONO with quadtree_mapping  for real-time dense mapping. How can I handle this?

WANG-KX, 2019-05-29:
Hi,

Thanks for your interest!
We have a version that supports VINS-MONO, and is currently using it and improving it. We will release it as soon as possible. Currently, I can release a middle-way code in the new branch and I cannot guarantee it is bugfree.

Regards,
Kaixuan

WANG-KX, 2019-05-29:
According to our experiments, the depth quality from MVDepthNet is better than that from open_quadtree_mapping. Still, you can try fusing open_quadtree_mapping.

closed 2019-05-29 by WANG-KX