HKUST-Aerial-Robotics/open_quadtree_mapping #24: How to get more depth estimates

wangyuanbiubiubiu, 2019-07-29:
When I used this program for depth estimation, only a small part of the pixels of my image were successfully estimated for depth, and I want to know which part of the threshold value can be adjusted to make more pixels for depth estimation. I don't need high precision, I just need more pixels for depth estimation

wangyuanbiubiubiu, 2019-07-29:
Another question is, can the program change the estimated depth range

WANG-KX, 2019-07-29:
1, The filter is to remove unstable depth estimations. If you want to disable the filter (which I highly disencourage you to do so), you can comment the [line](https://github.com/HKUST-Aerial-Robotics/open_quadtree_mapping/blob/353eb7baec43dabffa15b864a0e65019ffb75b5e/src/seed_matrix.cu#L165).
2, The range is hardcoded in some functions.

wangyuanbiubiubiu, 2019-07-29:
> 1, The filter is to remove unstable depth estimations. If you want to disable the filter (which I highly disencourage you to do so), you can comment the [...]

就注释掉这一行就够了是么,我现在就是重建出来的深度也太少了,我另一个问题是,这个程序是把所有的深度值都会显示出来么?有没有什么只估计50米以内的深度,因为我的场景深度很大,我怕这个程序有这样子的机制

WANG-KX, 2019-09-14:
重建的深度只在0.5~50.0m之间。
如果深度很大，算法可能无法三角化50m以上的场景。

ynma-hanvo, 2022-07-29:
> 重建的深度只在0.5~50.0m之间。 如果深度很大，算法可能无法三角化50m以上的场景。

does the algorithm works well if we limit range to 0 - 15m ,  
i tested it but seems it is still estimating depth for the values beyond 15m as  15m ?