HKUST-Aerial-Robotics/open_quadtree_mapping #26: Mapping

shermes91, 2019-09-13:
Hi,

I'm very new to this stuff, so sorry for maybe dumb questions. I don't know where the results for the depth map are saved. I have started the example launch and with the bag file and it worked. But I don't know how I can continue. Where are the results saved and how can i do a 3D construction of that. Can I use the OpenChisel for that or do I need the modified one?

WANG-KX, 2019-09-14:
Hi,
Thanks for your interest in the project.
The depth maps are not saved in your computer. If you would like to reconstruct the 3D map, you should use a fusion method that takes depth maps and camera poses and generate 3D models. OpenChisel is an option.

shermes91, 2019-09-14:
Thank you for the fast answer. So I have to start vins-mono, quadtree and OpenChisel at the same time? Can I follow the instruction of OpenChisel or do I have to modify it like you do?

Thanks again!

WANG-KX, 2019-09-14:
since the row topic type is not exactly the same between each component, some modifications are needed. You can use the open chisel under my github.

shermes91, 2019-09-14:
I will try that, thank you!

shermes91, 2019-09-14:
I've downloaded your OpenChisel. But catkin_make give me an error.

[ 96%] Linking CXX shared library /home/sebastian/catkin_ws/devel/lib/libchisel_ros.so
/usr/bin/ld: cannot find -lbfd
collect2: error: ld returned 1 exit status
OpenChisel/chisel_ros/CMakeFiles/chisel_ros.dir/build.make:312: recipe for target '/home/sebastian/catkin_ws/devel/lib/libchisel_ros.so' failed
make[2]: *** [/home/sebastian/catkin_ws/devel/lib/libchisel_ros.so] Error 1
CMakeFiles/Makefile2:2947: recipe for target 'OpenChisel/chisel_ros/CMakeFiles/chisel_ros.dir/all' failed
make[1]: *** [OpenChisel/chisel_ros/CMakeFiles/chisel_ros.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

[catkin_make.txt](https://github.com/HKUST-Aerial-Robotics/open_quadtree_mapping/files/3612720/catkin_make.txt)

WANG-KX, 2019-09-14:
Just remove -lbfd from the OpenChisel/chisel_ros/catkin.cmake

shermes91, 2019-09-15:
Thank you, so catkin works and I can start chisel_ros. In which order I have to start quadtree, vins and openchisel to display a map in rviz? Do you have an example?

WANG-KX, 2019-09-18:
Just open several terminals and run each node in the terminal. Or you can write a ROS launch file to launch them together. I do not have any script for examples since it has been a long time from the project.

shermes91, 2019-09-28:
Hi again,

I try to launch my VINS-Mono and Quadtree launch-file. But I get this error in my VINS-Mono launch:

[ERROR] [1569664952.528719766]: Client [/open_quadtree_mapping] wants topic /vins_estimator/camera_pose to have datatype/md5sum [geometry_msgs/PoseStamped/d3812c3cbc69362b77dc0b19b345f8f5], but our version has [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7]. Dropping connection.

Do you have an idea? Thx again!

WANG-KX, 2019-09-28:
Hi,

This is caused by the message difference between quadtree mapping and VINS. VINS publish the camera pose in nav_msgs/Odometry and quadtree mapping subscribes pose in geometry_msgs/PoseStamped. You need to change the code of quadtree mapping to subscribe the message type.

If you are not familiar with ROS, maybe I can change it when I am free next week.

shermes91, 2019-09-28:
I'm not familiar with ROS, so it will be great, if you can change it next week. But till then I will try to find it in the code. Thanks!

shermes91, 2019-09-28:
Hey, 

I could change the code of Quadtree and OpenChisel by myself and use Odometry now. I can now also display the full_mesh in rviz. Thanks a lot for your help!

closed 2019-09-28 by shermes91