HKUST-Aerial-Robotics/open_quadtree_mapping #27: Which GPU-based TSDF to use?

stevechan1993, 2019-10-15:
I got the full_mesh,and which gpu-based TSDF method can I use to support autonomous flight? Thanks a lot.

WANG-KX, 2019-10-15:
Sorry, I don't get your question. The map depends on your planning method. Also, why TSDF map sine you already get the meshes?

stevechan1993, 2019-10-15:
Maybe I missunderstand this sentence "We use a modified open chisel for 3D reconstruction and use a GPU-based TSDF to support autonomous flight". I want to control an A3 flight controller drone to autonomous flight, which planning method should I use, can I use full mesh directly?

WANG-KX, 2019-10-16:
In our applications (e.g., UAV autonomous flight), we use a gpu-TSDF to fuse depth maps and extract occupied grids (grids with zero tsdf values).

Meshes are only for visualization that is not extracted during the flight. If we want to reconstrcuct the scene in meshes, we use OpenChisel for fusion and mesh extraction.

Later, we developed [SurfelFusion](https://github.com/HKUST-Aerial-Robotics/DenseSurfelMapping) which can fuse depth maps and extract (disconnected) meshes in the same time.